EinsteinsWorkshop/BlocksCAD #1: auth.js referenced but missing

anomaly256, 2015-09-13:
It seems there's a file missing from the blockscad/ directory, auth.js.  If I try to build the 'core' with the reference to the file removed, the resulting application is a bit broken.  I've checked on several branches (standalone, development, master) and the result is the same in each - if I rebuild without this file I am unable to render or create projects

mattminuti, 2015-09-13:
Just pushed a new commit, try the development branch again - Blockscad.offline (in blockscad/blockscad.js) should be set to true, and the build script won't look for auth.js anymore. Should work now!

yoderjen, 2015-09-13:
I've checked in a fix for the development branch. In blockscad.js, line 35, the variable
Blockscad.offline = true
needs to be set to true. 
I also added the handler for the project-new menu option, and fixed the build script.
Let me know if it still isn't working.
Thanks!

closed 2015-09-13 by yoderjen

anomaly256, 2015-09-13:
Yep, seems to be working now, thanks!  :thumbsup:

Also thanks for making this open source, it's pretty awesome.

yoderjen, 2015-09-13:
One more hiccup you may notice is that the "text" block (under the "Advanced" category) won't work unless you are running Blockscad on a webserver.  The "notes-webserver-localhost-blockscad" file has directions for running a simple web server locally using Python.

anomaly256, 2015-09-13:
I have been running it using the python SimpleHTTPServer module, everything appears to be working now.  Should Blockscad.offline still be true if running under a web server?  Does this just turn off the register/login options or does it affect other things as well?  I haven't done any digging through the code yet to familiarize myself with it, it's late here.  (I intend to wrap this in a rails or node app to handle auth and add some project management type abilities and server side storage / design sharing eventually)

yoderjen, 2015-09-13:
The Blockscad.offline variable is what we use to hook into our own Appengine based storage/project management system (which we have not open sourced).  So it has nothing to do with running the code on a local webserver.  I should rename that variable :).
The text library we're using (opentype.js, another great open source project) loads fonts using an XmlHTTPRequest, which can't be run off the local filesystem.  That's why you need to run it from a webserver.