EinsteinsWorkshop/BlocksCAD #2: Internationalization

dbrgn, 2015-09-15:
This is fantastic, the power of OpenSCAD with the ease of Scratch!

I would love to teach 3D modelling to kids with BlocksCAD. Is it translatable already? If not, how could we get it there?

yoderjen, 2015-09-15:
It is not currently translatable, though it is on the to-do list.  Our block library is Blockly, which has internationalization support, and I am hoping we can extend that for BlocksCAD strings and custom blocks.  I can probably get to it sometime this fall.

dbrgn, 2015-09-15:
That would be great :) If you need help with translating the strings, let me know. I could do the German translation.

SebCanet, 2016-01-25:
Hi,
I'm a french STEM teacher and I konw BlocksCAD will be very useful in fablab, for maths and STEM teacher. But also I would like to know if you can add internationalisation like in the demo code (https://blockly-demo.appspot.com/static/demos/code/index.html). That's what I used for my Blockly@rduino.
When you will need french translation....

Thanks for your work.

mattminuti, 2016-01-27:
The very first version of BlocksCAD was basically that demo, and had language selection, I think English French and Mandarin, but no translations. If memory serves, we dropped the language stuff because Blockly had changed their translation system. Needs investigation into what system they've moved to, and then all those hard-coded strings can get pulled out, as they should be :)

SebCanet, 2016-02-18:
ok, if you need help....

SebCanet, 2016-02-21:
I showed your BlocksCAD in my town's Fablab and they think that's a great tool for newbies.
They can create STL work without working in Openscad.
One of my colleague strated to translate BlocksCAD, hope you will make it international soon.

SebCanet, 2016-04-17:
French translation finished : http://blockly.technologiescollege.fr/BlocksCAD

yoderjen, 2016-04-17:
Fantastique!

I need to get language switching enabled on our front page.

yoderjen, 2016-05-18:
language-switching is now (mostly) built in to BlocksCAD!

You can check it out on the main page (blockscad.einsteinsworkshop.com) - I used SebCanet's translation for most of the french strings.  

All strings that need translating are now in one place: 

BlocksCAD/blockscad/msg/json

This directory has a bunch of files of string:value pairs.  The strings from this file are used to populate the web page.

To update a current translation (only .en, english, and .fr, french so far) you would modify the existing lang.json file, and then run create_messages.py from the msg directory.  

To create a new translation, create a new .json file for that language by copying one of the current .json files and modifying the strings for the new language.  Build it by running create_messages.py from the parent directory (msg).

qqq.json contains "definitions" of what each string is for (where it is located on the page, what it is trying to accomplish), so refer to this file if you are trying to figure out the context for what you're translating.

Thanks SebCanet for the french translation!

SebCanet, 2016-05-18:
GREAT !!!!!!!!!!!!!!!!!!!!!!!
I post it everywhere !!!
Thansk a lot for this great tool very useful in FabLab too.

dbrgn, 2016-05-31:
By the way, it might be worthwhile to put the translation process on something like Transifex (http://docs.transifex.com/formats/json/) to make contributions easier.

closed 2018-05-01 by yoderjen