EinsteinsWorkshop/BlocksCAD #3: Rendering fails when defining a variable inside another block

konstin, 2015-09-15:
#### How to reproduce
 1. Create a new project
 2. Add a block that can contain other blocks, e.g. a "sides" block
 3. Add a variable-defining block with a variable and a numeric value inside the "sides" block
 4. Click render. The rendering will fail because of Syntax Error due to a missing semicolon

#### Example Blocks-xml-file
```xml
<xml xmlns="http://www.w3.org/1999/xhtml"><block type="$fn" id="1279" x="-83" y="67"><value name="SIDES"><block type="math_number" id="1280"><field name="NUM">8</field></block></value><statement name="A"><block type="variables_set" id="1327"><field name="VAR">some_var</field><value name="VALUE"><block type="math_number" id="1348"><field name="NUM">42</field></block></value><next><block type="sphere" id="1359"><value name="RAD"><block type="math_number" id="1360"><field name="NUM">10</field></block></value></block></next></block></statement></block></xml>
```

yoderjen, 2015-09-15:
Thanks.  This is a known issue and is a problem with our openscad parser.  Right now variables are only supposed to be defined globally or in a module definition, but we don't have error checking on the blocks yet to enforce that.

yoderjen, 2015-10-27:
I've disallowed variables to be attached where our parser cannot handle them until I can fix the parser.  I'll close this for now.

closed 2015-10-27 by yoderjen

konstin, 2015-10-27:
Thanks for fixing that!