EinsteinsWorkshop/BlocksCAD #4: Rendering failure with variable assignment in block

rcj4747, 2015-09-15:
I am trying to increment a variable based on a loop counter and hitting a rendering failure.  This might be the same as issue #3, but I can't be certain.  If that is the case, how should I perform this sort of variable modification for a loop?  I was trying to create a stack of increasingly larger spheres on the y-axis.

[image]

OpenSCAD code view:
```
//!OpenSCAD

offset = 0;
for (radius = [1 : abs(2) : 10]) {
  offset = offset + radius;
  translate([0, radius, 0]){
    sphere(r=radius);
  }
}
```

XML:
```
<xml xmlns="http://www.w3.org/1999/xhtml"><block type="variables_set" id="688" x="28" y="7"><field name="VAR">offset</field><value name="VALUE"><block type="math_number" id="705"><field name="NUM">0</field></block></value><next><block type="controls_for" id="784"><field name="VAR">radius</field><value name="FROM"><block type="math_number" id="785"><field name="NUM">1</field></block></value><value name="TO"><block type="math_number" id="786"><field name="NUM">10</field></block></value><value name="BY"><block type="math_number" id="787"><field name="NUM">2</field></block></value><statement name="DO"><block type="variables_set" id="710"><field name="VAR">offset</field><value name="VALUE"><block type="math_arithmetic" id="727"><field name="OP">ADD</field><value name="A"><block type="variables_get" id="732"><field name="VAR">offset</field></block></value><value name="B"><block type="variables_get" id="794"><field name="VAR">radius</field></block></value></block></value><next><block type="translate" id="755"><value name="XVAL"><block type="math_number" id="756"><field name="NUM">0</field></block></value><value name="YVAL"><block type="variables_get" id="763"><field name="VAR">radius</field></block></value><value name="ZVAL"><block type="math_number" id="758"><field name="NUM">0</field></block></value><statement name="A"><block type="sphere" id="774"><value name="RAD"><block type="variables_get" id="801"><field name="VAR">radius</field></block></value></block></statement></block></next></block></statement></block></next></block></xml>
```

yoderjen, 2015-09-15:
This is the same as issue number three.  
To make a line of growing spheres on the X axis, you could do this:

[image]

The spheres will just touch if you have your loop index count by one.

mattminuti, 2015-09-15:
Also, variable reassignment doesn't make sense in OpenSCAD, and will give you exactly the same result. There's lengthy discussions on their forums about whether it's a good or bad thing, but we're matching their behavior.
It sometimes helps to think of it more like a hardware description language than a "normal" procedural programming language.

Fixing #3 will prevent this confusion in the future :)

rcj4747, 2015-09-15:
Thank you @yoderjen for the example.
Thank you @mattminuti for the background, I was coming at this thinking like a software engineer.

I'll close this bug since it's a duplicate of #3.

closed 2015-09-15 by rcj4747