EinsteinsWorkshop/BlocksCAD #8: Orthogonal View

mattminuti, 2015-09-16:
It would be really nice to have an orthogonal view mode.

z3dev, 2017-06-15:
As in.... projection?

OpenSCAD has this functionality. Is this simular?

mattminuti, 2017-06-15:
Not as in OpenSCAD's projection(), though that's always been on the
wishlist, since it's super-duper useful. Didn't realize that it wasn't
listed in the issues list, or my original TODO file from forever ago,
probably was assuming we'd get it naturally from using OpenSCAD's engine,
which is no longer a current goal.

This issue was poorly named. I should have entitled it something more like
"Orthographic Projection View." The current view uses perspective, which is
undesirable in many contexts.