EinsteinsWorkshop/BlocksCAD #10: Unable to delete individual blocks on development branch

anomaly256, 2015-09-18:
After checking out 58a2fdb7856580b03b1aed6fada17ed26746da07 the dropdown menu for blocks only contains 'disable all blocks' and 'enable all blocks' and no longer shows the option to disable individual blocks nor delete them.

anomaly256, 2015-09-18:
Note: this only seems to happen if I rebuild anything, using the *_compressed.js files from the git repo seem to work ok

yoderjen, 2015-09-18:
Let me see if I can reproduce this problem.  I haven't seen it before.  Thanks!

yoderjen, 2015-09-18:
Okay, this is related to issue #9 - see my comment there.  Short form - don't change blockly/msg/json/en.json, undo your changes to blockly/msg/js/en.js, and this should fix the BlocksCAD problems.  I'll see if I can figure out why the Blockly message generation code isn't working.

yoderjen, 2015-09-18:
I fixed en.json in the latest commit to development (795047f).  This should solve your problems.  Thanks for pointing this out!

closed 2015-09-18 by yoderjen