EinsteinsWorkshop/BlocksCAD #9: blockly/msg/json/en.json parse error when building blockly components

anomaly256, 2015-09-18:
Heyas,

On my system I'm seeing a python error when running build.py in the blockly directory on the development branch (commit 58a2fdb7856580b03b1aed6fada17ed26746da07):

```
Error reading ./msg/json/en.json
Traceback (most recent call last):
  File "i18n/create_messages.py", line 149, in <module>
    main()
  File "i18n/create_messages.py", line 65, in main
    source_defs = read_json_file(os.path.join(os.curdir, args.source_lang_file))
  File "/home/anomaly/devel.hosting/BlocksCAD/blockly/i18n/common.py", line 64, in read_json_file
    raise InputError(filename, str(e))
common.InputError: ./msg/json/en.json: Expecting property name enclosed in double quotes: line 7 column 2 (char 173)
('Error running i18n/create_messages.py: ', CalledProcessError())
```
The python json parser doesn't seem to like the stray commas after the final properties in a list.  Changing this
```
{
        "@metadata": {
                "author": "Ellen Spertus <ellen.spertus@gmail.com>",
                "lastupdated": "2015-09-15 10:09:56.877000",
                "locale": "en",
                "messagedocumentation" : "qqq",
        },
}
```
to this
```
{
        "@metadata": {
                "author": "Ellen Spertus <ellen.spertus@gmail.com>",
                "lastupdated": "2015-09-15 10:09:56.877000",
                "locale": "en",
                "messagedocumentation" : "qqq"
        }
}
```
resolves the issue.  My system is using python 2.7.3 (Debian Wheezy)

yoderjen, 2015-09-18:
Thanks.  I'll pull that in.

yoderjen, 2015-09-18:
Okay, I won't pull that in without figuring out what is going on.  
When you make this change to en.json, Blockly ends up missing a lot of messages which is causing the problems in issue #10.  I need to track this down.  You need to get the original file from this branch for 
/blockly/msg/js/en.js
and you need to NOT change the en.json file for now.  Accept the errors in the Blockly build - it won't update the language files, but that's better than having BlocksCAD stop working.
The re-build after changing en.json is causing a lot of problems.  This is in the Blockly core, but is still probably my fault, I'll see if I can track it down.

anomaly256, 2015-09-18:
Ok, thanks

anomaly256, 2015-09-18:
Confirmed, ignoring this error fixes #10 

I can try a later python on a different (gentoo) host after work if you like, maybe it's an issue with the json module from the older python stdlib, or have you already tested that?

yoderjen, 2015-09-18:
I've tracked this down.  I fixed en.json, so that now the build process works and all the correct messages are being generated.  I'm closing this issue and #10.  Get the latest commit (795047f) and thanks for pointing this out.

closed 2015-09-18 by yoderjen

anomaly256, 2015-09-18:
Thanks for fixing it so fast :)