EinsteinsWorkshop/BlocksCAD #16: Non-manifold STL

KitWallace, 2015-10-12:
The simplest union of objects creates STL which RepetierHost judges to be non manifold.  Generally slic3r can handle these but not always, and cleanup via say netfabb would be tedious.  The equivalent code in OpenSCAD has no such problem eg.

cube(10);
translate([10,5,0]) cube(10);

KitWallace, 2015-10-17:
I should add that CuraEngine does manage to slice the STL on the examples I've tried.

yoderjen, 2015-10-17:
We've had reports of problems with the slicers RepetierHost and FlashPrint.  Cura, Simplify3D, and MakerBot slicers don't have problems.  We're going to be looking into how to fix this.

yoderjen, 2016-02-17:
This is a t-junction issue - our geometry engine, csg.js, creates meshes with t-junctions.  Some slicers handle this okay (as noted above).  You can also pass the .stl's through netfabb to clean them up.

z3dev, 2017-01-16:
FYI. There's a fix for this at the new JSCAD organization.

yoderjen, 2017-01-16:
I think you are referring to the "fixTJunctions()" function.  I have tried this, and it doesn't work except in trivial cases.

closed 2018-05-01 by yoderjen