EinsteinsWorkshop/BlocksCAD #18: Calculating with angles

thesourcerer8, 2015-10-17:
When I take a Rotation block, it has an Angle sub-block. If then then want to multiply that angle by a certain factor, I take a multiply block, insert it into the Rotation block, instead of the Angle block, and then take the Angle block that fell out into the Multiply block. Then I add a number block as the factor, or some variable block as the second multiplier. But this does not work, a multiplication only wants to operate on numbers, not on angles. 5*10°=50° does not seem to work. The workaround I have found so far is to use a number block instead of the Angle block, and write the degrees as a number in there. Then the multiplication works. The disadvantage is that I cannot graphically select the angle anymore.

yoderjen, 2015-10-17:
I have not been able reproduce this error.  In the code produced from the blocks (viewable from the "code" tab on the upper right), the angle block simply produces a number, and 5 * 10deg = 50.

[image]

rotate([(5 * 10), 0, 0]){
  cylinder(r1=5, r2=5, h=20, center=false);
}

Can you provide more details on what you are seeing?

thesourcerer8, 2015-10-17:
Great, it works. Yes. It seems to me this was solved already. Sorry, I should have tested it again, before reporting it.

closed 2015-10-17 by thesourcerer8