EinsteinsWorkshop/BlocksCAD #22: recursion not supported in modules

KitWallace, 2015-11-15:
recursive call in a module generates a syntax error even though the resultant OpenSCAD code executes correctly.

mattminuti, 2015-11-19:
Would you mind providing an example so we can track this down?

yoderjen, 2015-11-19:
Recursion sometimes works, and sometimes doesn't.  The problem is buried somewhere in our parser.

This recursive menger sponge works, with the loops directly inside the module def.:
[image]

while this rearrangement of the same code fails with a parsing error:

[image]

yoderjen, 2015-11-19:
So sometimes you can rearrange the code in a way to make recursion work.  I don't know exactly where in the parser the problem is at this time.  It is going on our long list of things to fix in the parser, thanks.

KitWallace, 2015-11-19:
This was my simple example:

[image]

mattminuti, 2015-11-19:
Adding an else condition with anything in it makes it work, so maybe the parser doesn't know how to handle the exit case...? Thanks for the example, it'll help!

yoderjen, 2017-01-09:
As of version 1.7.0, our parser is much more flexible in how you can code recursive modules.  If you have any examples that still don't work in 1.7.0, please let me know.

closed 2018-05-01 by yoderjen