EinsteinsWorkshop/BlocksCAD #24: Unable to set variables in module

KitWallace, 2015-12-15:
I want to compute local variables based on parameter values but attempts to add set variable blocks to a module are rejected although the obvious OpenSCAD code (with or without assign() ) is fine.

This is rather important when developing good programming practice.  Without it you end up with lots of meaningless duplicated numbers and expressions.

yoderjen, 2016-01-22:
this is "fixed" in the development branch.  The bug that disallowed variable declarations in a module is fixed.  We still don't match openscad's new parser's flexibility in setting variables.  That fix has to wait on the parser rewrite.
this should go out in the new release this week.  Thanks!

closed 2016-01-22 by yoderjen

yoderjen, 2016-02-17:
fix has finally been released.  Thanks for your patience.