EinsteinsWorkshop/BlocksCAD #28: Ubuntu Studio 15.10 localhost fails

CoolGames, 2016-04-14:
# Ubuntu Studio 15.10 using Chrome Version 48.0.2564.116 (64-bit) on Apache2 with
### Ubuntu Studio 15.10 localhost fails with ABORT not RENDER as online does

*Installed nodejs as required*

```
apt-get install nodejs
```

### Then clone and try

```
cd '/var/www/html'
git clone https://github.com/EinsteinsWorkshop/BlocksCAD.git
```

### Screen snapshot when 'http://localhost/BlocksCAD/' 
## or maybe this relates to displayed results
```
python -m SimpleHTTPServer 9000
Serving HTTP on 0.0.0.0 port 9000 ...

127.0.0.1 - - [14/Apr/2016 16:13:27] "GET /blockscad/lightgl.js HTTP/1.1" 200 -
127.0.0.1 - - [14/Apr/2016 16:13:27] code 404, message File not found

```

###  SimpleHTTPServer 'http://localhost:9000/'

[image]

yoderjen, 2016-04-14:
Are you cloning the master branch?

yoderjen, 2016-04-14:
Here's what you need to do:  

from your root directory, enter the "blockscad" directory, then find the file "blockscad.js".  Edit line 34:

Blockscad.offline = true;

(it is currently set at false and if you switch it you'll see the same behaviour as our production website except for no cloud backend)

CoolGames, 2016-04-15:
### Thanks that is easy.
I  copied path to clipboard where there is README.md on github then 

```
cd /var/www/html 
git clone https://github.com/EinsteinsWorkshop/BlocksCAD.git
```
.
## Then edit line 34 in /blockscad/blockscad.js set to true.

[image]

yoderjen, 2016-04-15:
Glad that worked for you.  I'm closing this issue.

closed 2016-04-15 by yoderjen