EinsteinsWorkshop/BlocksCAD #29: README allows localhost but fonts missing and setting offline with undocumented edit

CoolGames, 2016-04-28:
"Error: Failed to Load Font:" when using TEXT on localhost.
next issue is error
blocscad.js needs to be set offline true as revealed by phone call to developer
Blockscad.offline = true;  // true unless using a cloud service backend for file management

yoderjen, 2016-04-28:
You need to run a local webserver to be able to load fonts.  I usually use either a python webserver

python -m SimpleHTTPServer 9000

or the (faster!) node.js server

http-server -p 9000

then access the page at localhost:9000.

Does this fix your font loading issue?

CoolGames, 2016-04-28:
I am running full Apache2

mattminuti, 2016-04-28:
That's what we're using in production, a bog-standard Apache install.

CoolGames, 2016-04-28:
I am using both Ubuntu Studio and Beagle Bone Black with machinekit (LinuxCNC / EMC2) and neither is able to find the fonts in the subdirectory on BlocksCAD/fonts
Is there packages required for truetype on Linux ?

CoolGames, 2016-04-28:
Also uploaded to my GoDaddy domain in subdirectory using git clone and it also has the error

mattminuti, 2016-04-28:
I'll try to check this against a fresh Ubuntu image tonight to figure out
what dependencies are actually required. If memory serves, we're hosting on
Ubuntu, but it could be centos.

yoderjen, 2016-04-28:
silly things to check: 

1. are the font files actually in your filesystem?
2. if you run a node.js webserver, is it able to load the fonts?

CoolGames, 2016-04-28:
//!OpenSCAD

Not sure how to run node.js webserver separately ?
Fonts can be found in filesystem in BlocksCAD subdirectory and only ones offered !

Here is what I save from on line to localhost to get the error

translate([-50, 0, 10]){
  color([0.2,0.8,1]) {
    linear_extrude( height=2, twist=0, center=false){
      text("Cool Games", font = "Roboto", size = 20);
    }

  }
}

yoderjen, 2016-04-28:
if you have node.js installed, try this:

navigate to the top level directory of blockscad (it should contain index.html)
http-server -p 9000

if you don't have node.js, but you have python, from that same directory you can instead run
python -m SimpleHTTPServer 9000

then from your web browser, access BlocksCAD from http://localhost:9000

CoolGames, 2016-04-28:
Python worked so what is difference from apache running the code ?
Is there a need to reconfigure on localhost and GoDaddy ?

yoderjen, 2016-04-28:
No idea.  Hopefully Matt will have some ideas.  If you manage to fix it, let me know what needed to happen so I can help others in the future.

CoolGames, 2016-04-28:
I have nodejs 4.2.6 installed including dependencies with synaptic 

```
leetdavy@leetdavy-StudioAir:/var/www/html/BlocksCAD$ dpkg -l | grep node
ii  nodejs                                        4.2.6~dfsg-1ubuntu4                                 amd64        evented I/O for V8 javascript
leetdavy@leetdavy-StudioAir:/var/www/html/BlocksCAD$ 

```

yoderjen, 2016-05-18:
Closing this because I think it was not a BlocksCAD issue.

closed 2016-05-18 by yoderjen