EinsteinsWorkshop/BlocksCAD #30: BlocksCAD to OpenScad shifted text

CoolGames, 2016-04-28:
## Screen shots of 3D text on modified cube reveals a shift when save as SCAD is used.

STL(binary) appears correct in Mesh Lab.

### Notice second line is also shifted from G over N 

[image]
[image]
[image]

yoderjen, 2016-04-28:
Interesting.  I can see one Blockscad bug, which is that we should be reporting the font used as "Roboto:style=Bold" for openscad, but that doesn't fix the size issue.  I don't know whose font is the "right" size.  A size 20 font is clearly rendering much larger in openscad.  I'll look into this.

yoderjen, 2016-04-28:
Openscad font sizes are in points (that's what their font library uses), Blockscad's font sizes are in pixels (thats what our font library uses).  Bleah.

yoderjen, 2016-04-30:
I fixed the font size but forgot to fix the roboto style to be bold.  Oops.  Well, at least the size is fixed in the current production version.

closed 2016-04-30 by yoderjen