EinsteinsWorkshop/BlocksCAD #35: Generate Button

mskmoorthy, 2016-10-05:


yoderjen, 2016-10-05:
Does it work in firefox for you?  What happens if you refresh the Chrome page?

mskmoorthy, 2016-10-05:
Viewport does not show (along with generate button) in Firefox 49.0 - Refresh in Chrome does not help either

yoderjen, 2016-10-05:
Is this report for the website at blockscad.einsteinsworkshop.com,  or are you using the source to get your own copy of the page running?

mskmoorthy, 2016-10-05:
I am using the source code to get my own copy running. (Sorry for not explicitly writing it)

yoderjen, 2016-10-05:
Ah, that makes more sense.  I did break the generate button about a month ago.  The issue should be fixed in the latest development branch commit from about three weeks ago.  Do you have the latest version?

yoderjen, 2016-10-05:
I found the problem and checked in the fix.  Get the newest version of the development branch (should have just been updated) and it will fix your problem.  Thanks for the report!

closed 2016-10-05 by yoderjen

mskmoorthy, 2016-10-05:
Thanks - works Like a charm!