EinsteinsWorkshop/BlocksCAD #37: Collaboration with OpenJSCAD.org

z3dev, 2017-01-04:
It seams that this project is making some changes to common libraries which both BlocksCAD and OpenJSCAD.org are using. How about some collaboration on these libraries, which can benefit both projects.

z3dev, 2017-01-04:
Thanks! I started watching this repository for conversations, and pull requests. If you haven't already then you might want to start watching our repository at https://github.com/Spiritdude/OpenJSCAD.org

There's a major reorganization of the source happening now. This will allow easier integration into NODE based projects. And we hope that this will allow others to make further innovations to OpenJSCAD.org as well.

yoderjen, 2017-01-04:
Are you guys working on OpenJScad.org or are you working on a different product?

the openscad parser (openscad-openjscad-translator) is out of date and buggy as well.  I've been trying to debug it but haven't had any luck extending it for the new openscad language features.

z3dev, 2017-01-09:
OpenJSCAD.org is the URL

We are also setting up an organization to manage all the connected libraries. If interested then please let me know. You are welcome to use any of these libraries and contribute as well.

https://github.com/jscad

z3dev, 2017-01-15:
@yoderjen The new organization is live. The authors of CSG.js and openscad-openjscad-translator have agreed that these would both be better under a single organization.

Please consider joining. We would love to have any contributions from this project as well.

z3dev, 2017-03-19:
Wouldn't your changes be better as part of the JSCAD open library?
If you don't mind then I will start merging changes from the translator source into the same at JSCAD open library; https://github.com/jscad/openscad-openjscad-translator

z3dev, 2017-06-15:
So.... we have released 1.0.0 which should make integration easier.... we hope.

Let us know if you have any isssues.

closed 2018-05-01 by z3dev

WolfgangFahl, 2019-03-17:
see my comment at https://openjscad.nodebb.com/topic/8/cooperation-with-blockscad