EinsteinsWorkshop/BlocksCAD #39: how to generate language files? I want to translate the soft into Chinese

shytears, 2017-02-17:
I can use "js_to_json.py“to generate zh.json, and then modify it. At last, use "create_massges.py" to generate zh.js.
But, after that, there is no changes on the webside.
please tell me the right processes to generate language files.

yoderjen, 2017-02-17:
You need to submit the zh.json file as a pull request so I can add it to our code base, then I'll have to release it to the blockscad3d.com website.

shytears, 2017-02-18:
ok. thanks for you reply so rapidly! This is my first project on github.I will do this in these day.

ps: Anther question is that , I use 2D/3D text block and it didn't works. When i press render button, it turns into working and gray contiousely, as shown blow. Then I have to reflash the web side.
the running evironment is runing the website localy , with the index.html in the directory, didn't pulic it with the web service.
[image]

yoderjen, 2017-02-18:
The fonts aren't being loaded.  The font library we use (opentype.js) needs a web server to load the fonts.  So when you are running code locally, you'll need to run a web server, which is easy.  From the root directory (with index.html in it), you can run an http server - I recommend either using Node.js's server or Python's.

The node server command is http-server -p 8888  (use whatever number you want the port to be).  You'd then go to

localhost:8888  

in the browser to see the page.

read the README for the instructions on how to run the python simplehttpserver.

shytears, 2017-03-05:
I've sent a pull request for Chinese language file.
Because it's the first time I use pull request, I have sent a zipped zh.json file. I hope that I have done a right thing.

closed 2017-03-07 by shytears