EinsteinsWorkshop/BlocksCAD #44: render error in line40 [uncaught DOMException]

shytears, 2017-04-14:
[image]
sometimes happened to me in Chinese languege.

yoderjen, 2017-04-14:
Oops.  Can you give me any details of when this happens?  Send me the code, or a full screenshot?  Does it only happen when rendering, or some other time?

thanks

yoderjen, 2017-04-14:
Actually, since it is in Chinese, you'll probably need to send me the xml code for what causes the error.  Save the blocks to your computer and hopefully you can attach them here.

yoderjen, 2017-04-14:
Or you could share the project and give me a link to the project on blockscad3d.com for me to check out.

yoderjen, 2017-04-14:
I see another problem too.  None of our fonts (in the Text blocks) can do chinese.  Do you have any recommendation for a free chinese font?  I like to choose thick (bold, or heavy) fonts because they work the best on 3D printers.

shytears, 2017-05-11:
I'm sorry so long to reply you.
For the first question, I found that maybe it was caused by browser.At that time, I used Tencent brower which based on chrome with deep modify. Then I change to chrome or Tencent for new version, the problem is gone.

closed 2017-05-11 by shytears

yoderjen, 2017-05-11:
That's okay, I'm glad it's working, because I haven't been able to reproduce that error.  

I had another user contribute more Chinese for the translation, and I'm working on a Chinese font (so that you can model simplified Chinese characters).

thanks!

shytears, 2017-05-11:
I have no idea for the free font for BlocksCAD.
BUT I found another problem maybe caused by fonts of browser.
When I use my WIN10 notebook, with no matter Tencent browser or Chrome, the languege would look normle in English or Chinese simple..BUT when I use rasperryPi with chrome or firefox, I found that I can only see Chinese version by open BlocksCAD directly in folder, not in the the sever mode. shown as below:
<img>
[image] 
</img>