EinsteinsWorkshop/BlocksCAD #45: Model is not on a closed surface

osupk8, 2017-07-09:
Hi,

I want to point a bug I found using a model I created today : https://www.blockscad3d.com/community/projects/43698

When I tried to export this model in STL format to my 3D printer I got the message : "Model is not on a closed surface"
And when I try to print this model it fails. I also have trouble when i try to import this STL model in tinkercad,etc it bugs.
Perhaps the problem  is in STL générator ?

Indedd, I take the code of this model and I copy/paste it  in opensCAD software and when I export the model in STL it works fine on my printer.

Best regards

Christophe

yoderjen, 2017-07-09:
What slicer/printer are you using?

osupk8, 2017-07-09:
Dremel3D slicer

----- Mail original -----

yoderjen, 2017-07-31:
Added some mesh repair before STL download.  It doesn't fix all models, but is a start.

closed 2017-07-31 by yoderjen